PugachA/UnmannedAerialVehicle #16: Реализовать СИ ВСП

Li2s, 2019-11-04:
- [ ] Считать данные с MS5611
- [ ] Реализовать фильтр для получения вертикальной скорости
- [ ] Считать данные с аналогового ПВД
- [ ] Реализовать алгоритм получения углов атаки и скольжения с 5 ПВД

Li2s, 2019-11-05:
@NoMonsters @PugachA прошу вас оценить реализацию опроса MS5611 в классе, представленную в ветке feature/add-MS5611-lib

PugachA, 2019-11-05:
Никит, так не делается. Нужно создать PR, чтобы нам было удобно проверять изменения. Был удобный инструмент сравнения. И в PR можно ставить Апрув или реджект

Li2s, 2019-11-05:
@PugachA это не пойдет даже в девелоп, поскольку плата поменяется, только поэтому не стал создавать PR

NoMonsters, 2019-11-06:
Фильтры реализовывать по #21

NoMonsters, 2020-01-20:
@Li2s  Тут открыть отдельно issue на флюгарку и на ПВД для скоростного напора эту issue закрыть.

closed 2020-02-16 by Li2s