PugachA/UnmannedAerialVehicle #23: Сборка в Keil

NoMonsters, 2019-11-10:
Можно ли в Keil отделать папку со всем этим барахлом от папки с исходниками? Наверняка можно. Выши пулл реквесты невозможно ревьюить. Я буду их заворачивать пока тут не будет ответа, возможно или нет.

Li2s, 2019-11-10:
Я не понял, что конкретно ты хочешь

NoMonsters, 2019-11-10:
Я хочу, чтобы была папка в которой меняются только исходники проекта. И не было тыщу файлов .hueprjft .nikitalitus и прочих. Даже когда Саня сделал PR уже с добавленым барахлом, всё равно при просмотре разницы есть ещё 13 системных файлов. Это супер неудобно. И таким образом я хочу, чтобы если вы сделали изменение только в main.c в PR я видел только main.c

Li2s, 2019-11-10:
В таком случае нужно отцеплять проект в Keil от контроля версий, как это было с машиной

NoMonsters, 2019-11-10:
Т.е. там нет никаких средств, которые позволяют как-то отделить исходники?

Li2s, 2019-11-10:
Кстати, если в pr выбирать только интересующий тебя коммит (как правило последний), то он не будет отображать все файлы, только измененные и соответствующие им бинарники (которые и не отображаются)

PugachA, 2019-11-10:
Если посмотреть мои последние коммиты, там меняется только `.с` или `.срр` файл, ну и файл проекта `.foton`. В принципе достаточно читаемо. Я пытался засунуть его в ингор, но какого хрена он все равно отображается

NoMonsters, 2019-11-11:
Понятно, смотрим диф по коммитам

closed 2019-11-11 by NoMonsters