PugachA/UnmannedAerialVehicle #40: Система отключения двигателя

NoMonsters, 2020-03-17:
Начал изучать вопрос.
Используем полевой транзистор + драйвер. Драйвер нужен т.к. у транзистора управляющее напряжение выше логического на ногах MCU.
Примеры транзисторов: [тут](https://www.chipdip.ru/product/auirfs8408-7trl)
Пример драйвера: [тут](https://www.chipdip.ru/product1/8002024653) и [тут](https://www.chipdip.ru/product/ucc37321p)

Полезные ссылки:

- [ссылка1](http://easyelectronics.ru/upravlenie-moshhnoj-nagruzkoj-postoyannogo-toka-chast-3.html) основные моменты подключения полевиков
- [ссылка2](http://electrik.info/main/praktika/1388-polevye-tranzistory.html) хороший пример с моделированием
- [ссылка3](http://easyelectronics.ru/vklyuchit-vyklyuchit-sxemy-upravleniya-pitaniem.html) схема управления питанием с MCU

NoMonsters, 2020-03-21:
Нет смысла коммутировать транзистором аккумулятор, т.к. у нас уже есть коммутатор такого тока- регуль двигателя. И чтобы отключить движок нужно просто сигнальный провод разомкнуть. А если сам регуль наебнется, то уж точно отрубится. Он прям 0.0000001% наебнется так, что выйдет просто на 100%

NoMonsters, 2020-03-22:
У нашего регулятора задержка между отключением сигнального провода и отключением движка - 5 **секунд**

NoMonsters, 2020-03-25:
[Первый вариант подхода]
[неактуально]

Начал изучать вопрос.
Используем полевой транзистор + драйвер. Драйвер нужен т.к. у транзистора управляющее напряжение выше логического на ногах MCU.
Примеры транзисторов: тут
Пример драйвера: тут и тут

Полезные ссылки:

    ссылка1 основные моменты подключения полевиков
    ссылка2 хороший пример с моделированием
    ссылка3 схема управления питанием с MCU

NoMonsters, 2020-10-01:
нах ты закрыл это же доки?! Это лишний раз доказывает, что хранит документацию в ишаках так себе идея

Li2s, 2020-10-17:
Перенес на вики https://github.com/PugachA/UnmannedAerialVehicle/wiki/%5BERS%5D-Система-отключения-двигателя

closed 2020-10-17 by Li2s