PugachA/UnmannedAerialVehicle #42: ERS потеря связи с основным контроллером

NoMonsters, 2020-04-03:
**Краткое описание:** Система должна выпустить парашют при потере связи с основным вычислителем

**Предусловия**: Сигнал от основного вычислителя потерян более чем на 0.1 с (изменился/исказился т.е. стал невалидным)

**Основной поток:**
1. Резервная плата переключает ШИМ от основного контроллера, на свой ШИМ, который соответствует 0 оборотов двигателя
2. Двигатель останавливается
3. Резервная плата отправляет на сервопривод сигнал на открытие люка
4. Сервопривод открывает люк

**Постусловия:**
Парашют выпущен

**Технические детали:**
 - Используем дополнительный ШИМ резервного контроллера, скважность которого соответствует 0 об/мин скорости вращения двигателя
 - Для переключения сигнала с ШИМ основного контроллера на ШИМ резервного контроллера используем мультиплексор
 - Используем доп. ШИМ, т.к. если просто положить сигнальный провод на 0, двигатель останавливается слишком долго - 5 секунд

NoMonsters, 2020-12-01:
@PugachA 
А у нас есть такое вообще?)
Вот все рофлили над этими требованиями а про это уже забыли вообще

NoMonsters, 2021-03-08:
@PugachA Будешь это делать? Если нет, то закрывай ишак

PugachA, 2021-03-09:
Это зависит, от того получится ли сделать стабильное раскрывание парашюта. Пока оставим