PugachA/UnmannedAerialVehicle #60: Апдейт библиотек Servo и RCChannel

Li2s, 2020-09-08:
Я думаю, что удобней на выходе с RCChannel иметь отклонение стика в процентах (условных единицах), а на вход Servo подавать градусы отклонения сервы. Это упростит работу с алгоритмами и исключит ошибки при пересчете из и в микросекунды.

NoMonsters, 2020-10-02:
@Li2s Это нужно для стабилизации?

Li2s, 2020-10-02:
> @Li2s Это нужно для стабилизации?

Да

PugachA, 2020-10-10:
@Li2s Т. е. RcChannel на выходе будет давать от 0% до 100% и Servo принимать от 0% до 100%. Или например и там и там использовать интервал 0-180 градусов?

Li2s, 2020-10-11:
@PugachA Я думаю, что RcChannel должен выдавать 0-100% по газу и -100 - 100% по остальным каналам. Потом будет какой-то блок перевода этих процентов в заданные значения, в зависимости от режима (вертикальной скорости, угловой скорости и тд). А Servo на вход будет принимать по газу те же 0-100%, а по остальным каналам - угол отклонения соответствующей поверхности.

closed 2020-11-07 by PugachA