PugachA/UnmannedAerialVehicle #97: Предоставить аппаратуру видео фиксации с возможностью воспроизведения на ПК

PugachA, 2020-10-23:
@NoMonsters  ты говорил, что у тебя есть камера для передачи видео сигнала. Можешь еще раз посмотреть, все ли у тебя есть для передачи сигнал на ПК, если нет то что нужно докупить.
Было здорово иметь две камеры. Чтобы больше объектом могли просматривать

NoMonsters, 2020-10-23:
Для компа нужен easycap. Чтобы AV перевести в цифру
Видео передатчик есть.
Камера есть, вроде бы даже 2. 
Нет свитча, чтобы между ними переключаться. источник для видео передатчика нет возможности выбрать. Может как-то на мультиплексоре бы это можно было замутить)))

NoMonsters, 2020-10-23:
Не особо виду применения этому. Зачем нам на что-то там смотреть риалтайм

PugachA, 2020-10-23:
Ну мне кажется полезно смотреть на рули и например флюгарку в realtime или нет?

PugachA, 2020-10-23:
Или например когда полет по маршруту, будет полезно смотреть что с происходит с самолетом.

PugachA, 2020-10-23:
Можешь поискать и заказать недостающую аппаратуру?

NoMonsters, 2020-10-28:
Себе: добавить видео часть, с запиской от отдельного акка

NoMonsters, 2020-11-13:
Заказать:
1. Easycap или аналог