PugachA/UnmannedAerialVehicle #102: Согласование ШИМов разных плат

NoMonsters, 2020-10-25:
Почему-то ШИМ который генер т плата сас крашит все ШИМ сигналы

PugachA, 2020-10-26:
Предлагаю попробовать:

- [ ] Запитать сервы отдельно от двигателя
- [ ] Сделать идентичные настройки ШИМ на двух платах. Поднять частоту.

@NoMonsters  Что еще можно?

NoMonsters, 2020-10-28:
Мб:

- [ ] Тебе использовать RcChannel, а не PWMCapture

- [ ] Использовать Servo с микросекундами а не градусы

PugachA, 2020-11-01:
Отловили ошибка, когда полностью умирает плата САС. Это происходит при большой нагрузке на сеть питания 3.3В, напряжение проседает и плате не хватает питания. Поднята напряжение на стабилизаторе и ошибка устранена

PugachA, 2020-11-01:
Нужно отловить кейс, когда появляются задержки на всех сервах

closed 2020-11-01 by PugachA