PugachA/UnmannedAerialVehicle #119: Добавить TrimReset на одну из кнопок пульта

Li2s, 2020-11-08:
Избавляемся от следующего кейса:
1) Летим вручную, оттримировали все каналы, средние значения уехали с 1500 мес
2) Включаем стабилизацию, маппинг мкс ШИМ->заданная угловая скорость сделан из расчета, что 1500 мкс-> 0 гр/с. При нулевом положении ручке заданная угловая скорость не ноль.

NoMonsters, 2020-11-13:
Решили, что вся логика будет на пульте, посмотреть что в OpenTx можно по этому поводу. Желательно, чтобы при возврате в ручной режим триммеры вернулись в установленные ранее пилотом значения.
Во всех режимах кроме ручного тримы в 0

NoMonsters, 2020-11-16:
Добавил обнуление тримеров в пульте используя flight mode. При положении тумблера соответствующего стабилизации триммеры обнуляются и недоступны, при других положениях стиков они возвращаются в старые значения

closed 2020-11-16 by NoMonsters