PugachA/UnmannedAerialVehicle #156: Сделать рилтайм рисовалку логов

NoMonsters, 2021-03-08:
Надо ещё как-то обозвать её поумному чтоб нам было понятно о чем речь

PugachA, 2021-03-09:
Наземная станция
Визуализация логов
PFD

NoMonsters, 2021-03-15:
@PugachA 
К Сб/Вс нужно:
Сформируем мб пока строку специально для настройки кефов:
например: "омега_зад, омега_тек, Vy_zad, Vy_тек, k_омега, I_омега, k_vy"
Нужно, чтобы совместно на графике отображались омеги или вертикальные скорости и где-то ещё числовые значения кэфов, которые мы подбираем, не графиком.

NoMonsters, 2021-03-15:
Ещё было бы прикольно передавать сигнал с канала режимов и согласно #157 отображать в каком режиме мы находимся, чтобы не путаться

NoMonsters, 2021-03-19:
Формат:
current_time;10*omega_x_zad;10*omega_x_tek;100*vy_zad;100*vy_tek;10*k_p_omega;100*k_i_omega;*10k_p_vy;mode_switch_pwm

Множитель перед параметрами - на что они домножаются чтобы в строке были инты при передаче, т.е. при отображении параметр надо разделить на соответствующий множитель

Отображение:
экран1: 

1. График текущей и заданной омеги на одном плоттере
2. Где-то рядом числовые значения кэфов k_p_omega и k_i_omega
3. Где-то ещё рядом поле с названием режима полета #157 описаны диапазоны. Дополню табличку названиями режимов сегодня поздно

экран2:
1. График текущей и заданной вертикальной скорости на одном плоттере
2. Где-то рядом числовые значения кэфа k_p_vy
3. аналогично экарн1

PugachA, 2021-04-03:
Сделано

closed 2021-04-03 by PugachA